squirrel-project/squirrel_driver #109: Wrist safety node does not reset

tpatten, 2017-05-29:
When the wrist safety node is triggered it does not reset in the same way as the bumber and airskin. Could it be implemented similarly so that we do not have to manually reset the wrist. Also, the [while loop (line 41)](https://github.com/squirrel-project/squirrel_driver/blob/indigo_dev/wrist_safety/src/wrist_safety.cpp) should be
`while(ros::ok()){
...
}`
not
`while(1){
...
}`
to ensure that the node is correctly terminated from the terminal.

ipa-nhg, 2017-05-29:
@JelizavetaK could you please take a look at this issue?

JelizavetaK, 2017-05-30:
Hi @ipa-nhg, maybe @Senka2112  can have a look at this, as she has developed the node.

ipa-nhg, 2017-07-31:
@Senka2112 could you please check this?

lokalmatador, 2017-08-30:
Well, it does after manually resetting; we also stumbled on this issue yesterday. It works but for sure could be way more convenient and elaborate.