squirrel-project/squirrel_driver #121: Hardcoded path in dynamixel sdk

federico-b, 2017-07-28:
Hi all,

in [dynamixel_tool.cpp line 97](https://github.com/squirrel-project/squirrel_driver/blob/indigo_dev/squirrel_control/src/dynamixel_sdk/dynamixel_tool.cpp#L97) the path to the model_info.list is hardcoded. However, the standard robotino setup does not install anything in that path. We should consider change that path to the [local resources in this package](https://github.com/squirrel-project/squirrel_driver/tree/indigo_dev/squirrel_control/dynamixel/models).

ipa-nhg, 2017-07-28:
@lokalmatador could you please fix this?

lokalmatador, 2017-07-28:
Well, not within the next two weeks - I'm leaving in an hour for holidays. You could ask Simon Haller, he also knows the problem. We originally were not able to export this file within the cmake build (sth failed). We thus temporarily moved it there. If you know how to do the former, please go ahead :) I'd be happy. Otherwise, in the meantime you can just copy the file to the respective location and we can fix it when I'm back. Ugly solution but at least only temporarily. Sorry for the delay.

lokalmatador, 2017-07-28:
@smhaller can you maybe fix this when you're back? Seems we totally forgot about it :D

lokalmatador, 2017-08-18:
Fixed.

ipa-nhg, 2017-08-21:
@lokalmatador could you please open a Pull Request?

lokalmatador, 2017-08-21:
should be there now

ipa-nhg, 2017-08-30:
can be closed as soon as https://github.com/squirrel-project/squirrel_driver/pull/133 is merged

lokalmatador, 2017-08-30:
Can't close it :(

ipa-nhg, 2017-09-20:
Fixed by https://github.com/squirrel-project/squirrel_driver/pull/133 , can be closed

closed 2017-09-20 by ipa-nhg