squirrel-project/squirrel_driver #122: `arm` namespace in squirrel_control.launch

federico-b, 2017-07-31:
Hi,

is the [`arm`](https://github.com/squirrel-project/squirrel_driver/blob/indigo_dev/squirrel_control/launch/squirrel_control.launch#L38) namespace necessary in the launch file?

ipa-nhg, 2017-07-31:
First of all the squirrel_control.launch is a temporal file to test only the arm. We will have to integrate this with the rest of the robotino file (camera, hokuyo..) and the shell components.

The relay of the joint_states was there only to test and check the rate of the topic, when we also have the wheels, the pan/tilt motors and the hand on the same launch file we will have a global joint states publisher for all of them.

Of course the namespaces are not necessaries but I like to use them to organize the software, all the parameters and topics of a component are always under the same path and makes much easier the debug of errors, at least for me....

federico-b, 2017-07-31:
Ok, no problems. Fine by me.

closed 2017-08-30 by ipa-nhg