squirrel-project/squirrel_driver #123: Base control unexpected behaviours

federico-b, 2017-07-31:
Dear all,

I noticed that `/squirrel_control_node` sometimes does not release the control of the base (and probably of the arm too). I also experienced this straight after booting the robot. For instance, it happens to me that I could not reposition the robot with the joystick cause the low level controller moved the robot to pose where the robot was switched on. 

I think `/squirrel_control_node` should execute the control loop if and only if requested. It might be also useful to create a service that kills the control loop just in case the a controller has been called via topic `/trajectory_trajectory_controller/command` and without the action server/client.

mzillich, 2017-08-01:
Is this not, why we had the twist multiplexer?

ipa-nhg, 2017-08-01:
are you starting a teleop node for ROS or the festo upstart control is controlling the joystick?

federico-b, 2017-08-01:
So that happed to me a couple of time. I joystick the robot using the low level driver by FESTO, no ROS teleop. What I noticed is that something was publishing `/cmd_rotatory`.

lokalmatador, 2017-08-18:
So what's the issue then here? Can we be more precise?

ipa-nhg, 2017-10-16:
was this solved?

lokalmatador, 2017-10-16:
I guess so, yes. @smhaller and @philkark was this finally fixed?

closed 2017-10-16 by ipa-nhg