squirrel-project/squirrel_driver #124: Reference frame of the controller

federico-b, 2017-07-31:
Dear all,

I noticed that the controller sometimes does not properly track the base trajectory. This might be due to the fact that the control is not computed in `/map`  frame but rather in the `/origin` or `/odom` frame. If this is the case, this is generally wrong because the transformation between `/map` and `/origin` (or `/odom`) are not constant (that's what the localizer is computing) and therefore the controller is affected by the errors/drifts in the odometry (which is not negligible).

Best

ipa-nhg, 2017-07-31:
I think this is exactly the same issue that we already had last year that was solved reseating the odometry :pensive: , I also think that a planner for a base should use the map frame, but then will be needed some localization software and we will mix manipulation and navigation functions, I think that the manipulation and navigation team should decide how to process together @mzillich  what do you think?

And pleaseeeeeeeeee in case we need to use map I recommend a separation of low level drivers of the arm (only move and get joint states of the arm) and the controllers for the 8dof, start a map server just to move one joint of the arm is really undesirable.

federico-b, 2017-07-31:
I think that controlling the base without using the localization will never work cause the feedback provided by the encoders will accumulate error on time. The point is that so far the base controller was implemented in squirrel_navigation and was independent of drivers and so on...

mzillich, 2017-08-01:
I agree with @federico-b. This should be in map frame. If we want to run the arm controller stand-along, then we just have to add a static transform map-odom, right?

federico-b, 2017-08-01:
Adding a static transform `/map` -> `/odom` when no localizer is running might also be a solution. What are we currently doing on our robot, we publish a static zero transform `/odom` -> `/origin` to connect the all transformation tree.

lokalmatador, 2017-08-18:
Again, what's the issue now? The controller computes nothing - it just sends the values to the motor or the base. In any case, I guess it's the planner's job to plan in the proper frame.

federico-b, 2017-08-28:
Hi,
so I might have misunderstood the role of the whole thing. But it seemed to me that the new controller was able to command the base to follow some waypoints... so there must be a trajectory controller for the base too... is that `ros_control_controller_manager`? Cause I guess that package is designed to control the arms of the PR2 (cause `move_base` controls the base), so it might be that it is designed assuming precise encoders (such as the one for the arms) and not for encoders with big noise, like odometry... but I might be wrong here, don't know much about `ros_control`. Anyway, if we use this controller only for few centimeters, I guess it should work anyway.

mzillich, 2017-08-28:
Yes, that's the plan to stay on the safe side: only use the 8DOF controller for short base movements. Long ones use the normal navigation services.

lokalmatador, 2017-08-28:
That depends on how many controllers you implement - one controller only works for a specific setup. So if needed, for sure, one can provide a ros_controller just for the base. The question is just how much sense this makes. I think we need to clarify all those things in our weekly hangout on Wednesday as there are apparently still some open issues on misunderstandings regarding needed and currently implemented functionality. Talk to you then. And good luck with the arm. If plugging in and out does not help, kick it fiercely ;)