squirrel-project/squirrel_driver #127: reversed joint directions

mzillich, 2017-08-02:
some joint directions seem to be reversed. I guess this is a mismatch between motor driving directions and joint directions as defined in the URDF.

lokalmatador, 2017-08-18:
Is this an issue of all robots? I remember that only ALU-FR reported this? So if this is the case, I guess this is related to the model and not the controller. If the controller works for two out of three robots, I guess the third one has issues with its model. As I have no clue - and also claim that it makes no sense to do such changes in the controller code - I suggest trying to fix the model. @ipa-nhg ?

ipa-nhg, 2017-08-18:
fix the model? what do you mean? if I change the URDF then UIBK and TUW will not work. 

From my point of view, there are two options:

- add the parameters motor directions on the controllers.yaml, modify the driver to read these parameters and create a configuration file per robot.

- Change the motor directions on the dynamixel firmware

If it is possible I vote for the second one!

lokalmatador, 2017-08-18:
Oh snap, you're right. Same base models. Well, then I'm with you and suggest that ALU-FR fixes this on the motor side and not on the controller side; the latter is just another dirty hack then.