squirrel-project/squirrel_driver #128: stop trajectory execution if obstacle

mzillich, 2017-08-02:
Can we at least stop the execution of the 8 DOF trajectory when an obstacle enters the planned trajectory (to then have the top level planner replan?

philkark, 2017-08-03:
That is really not a task for the planner as the plan has already been sent to the controller by that time and is completely disconnected from the execution. It would rather make sense that there is a cancel option in Philipp Zech's controller instead. We have found a topic in the new controller that seems to provide an action server inkl. a cancel option, but when Federico and I tried it out the robot did not stop the execution.

mzillich, 2017-08-04:
OK, that would be an additional safety input then.
All in all that is not trivial, as one would have to check the remaining trajectory for collision with the current octomap, in realtime. So I guess for now we have to rely on the skin to stop the robot when a contact occurs.