squirrel-project/squirrel_driver #129: stop publishing to cmd_vel

mzillich, 2017-08-02:
Currently squirrel_controller publishes commands to the joints and base constantly, as part of the SquirrelHWControlLoop::update() loop. It should hovever only send commands while executing a trajectory. Otherwise the squirrel_controller will "freeze" the base, and prevent move_base and even the joystick from moving the base.
So a flag or something "executing_trajectgory" would need to be added.

lokalmatador, 2017-08-18:
We'll have a look at it today.

lokalmatador, 2017-08-18:
Fixed. The controller now only publishes to /cmd_rotatory upon receiving a new trajectory; after the goal has been reached, nothing is published to the base anymore from the controller side (not the arm!). This is a dirty hack that somehow violates ros_control.

Please close.

ipa-nhg, 2017-08-21:
@lokalmatador thanks! could you please open a Pull Request?

ipa-nhg, 2017-08-30:
This can be closed as soon as https://github.com/squirrel-project/squirrel_driver/pull/133 is merged

lokalmatador, 2017-08-30:
Can't close :(

ipa-nhg, 2017-09-20:
merged https://github.com/squirrel-project/squirrel_driver/pull/133 can be closed

closed 2017-09-20 by ipa-nhg