squirrel-project/squirrel_driver #136: Unable to use robotino after the latest hardware/software changes

bajo, 2017-09-21:
@ipa-nhg @mzillich @edith-langer @tpatten @jibweb

We received the modified robot and tried to use it and found a lot of issues. 

- Raspberry pi not in the correct network -> unable to connect
- Raspberry pi not configured for ROS with multiple machines (ROS_MASTER_URI, etc. misconfigured) 
- Wifi does not autostart on robotino's internal pc (needs manual ifdown wlan0 / ifup wlan0)
- Unable to start `roslaunch robotino_bringup tuw-robotino2.launch ` from any user except squirrel
Error is `roslaunch robotino_bringup tuw-robotino2.launch 
... logging to /home/bajo/.ros/log/668f2dfc-9ec6-11e7-bfb3-00190f153f00/roslaunch-robotino-16659.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/indigo/share/xacro/xacro.py", line 62, in <module>
    xacro.main()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 696, in main
    eval_self_contained(doc)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 626, in eval_self_contained
    eval_all(doc.documentElement, macros, symbols)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 552, in eval_all
    (",".join(params), str(node.tagName)))
xacro.XacroException: Parameters [has_8dof] were not set for macro xacro:base
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/indigo/share/xacro/xacro.py '/home/bajo/catkin_ws/src/squirrel_robotino/robotino_bringup/robots/tuw-robotino2/urdf/robotino.urdf.xacro'] returned with code [1]. 

Param xml is <param command="$(find xacro)/xacro.py '$(find robotino_bringup)/robots/$(arg robot)/urdf/robotino.urdf.xacro'" name="robot_description"/>
The traceback for the exception was written to the log file
`
- Unable to start the new arm controller `roslaunch squirrel_control squirrel_control.launch`
Error is `ROS_MASTER_URI=http://robotino:11311

core service [/rosout] found
process[robotino_node-1]: started with pid [16797]
process[robotino_safety_node-2]: started with pid [16798]
process[robotino_odometry_node-3]: started with pid [16799]
process[twist_mux-4]: started with pid [16815]
process[twist_marker-5]: started with pid [16832]
process[arm_controller/squirrel_control_node-6]: started with pid [16855]
process[arm_controller/ros_control_controller_manager-7]: started with pid [16863]
process[arm_controller/robot_state_publisher-8]: started with pid [16880]
[ INFO] [1506001035.005349583]: Topic handler 'teleop' subscribed to topic 'cmd_teleop': timeout = 0.250000s, priority = 100
process[arm_controller/joint_states_relay-9]: started with pid [16888]
[ INFO] [1506001035.048606554]: Topic handler 'navigation' subscribed to topic 'cmd_navigation': timeout = 0.250000s, priority = 90
[ INFO] [1506001035.079042288]: Odometry connected to Robotino.
[ INFO] [1506001035.094627644]: Topic handler 'pushing' subscribed to topic 'cmd_pushing': timeout = 0.250000s, priority = 80
[ INFO] [1506001035.119603430]: Topic handler 'rotatory' subscribed to topic 'cmd_rotatory': timeout = 0.500000s, priority = 70
[ INFO] [1506001035.140738376]: Topic handler 'simulation' subscribed to topic 'cmd_sim': timeout = 0.500000s, priority = 1
[ INFO] [1506001035.160900205]: Topic handler 'pause_navigation' subscribed to topic 'twist_mux/locks/pause_navigation': timeout = None, priority = 101
[ INFO] [1506001035.187610348]: Topic handler 'pause_pushing' subscribed to topic 'twist_mux/locks/pause_pushing': timeout = None, priority = 91
[ INFO] [1506001035.194989512]: Topic handler 'pause_rotatory' subscribed to topic 'twist_mux/locks/pause_rotatory': timeout = None, priority = 81
[ INFO] [1506001035.199064080]: RobotinoNode connected to Robotino.
[ INFO] [1506001035.200156496]: Topic handler 'pause_sim' subscribed to topic 'twist_mux/locks/pause_rotatory': timeout = None, priority = 1
[ INFO] [1506001036.124309038]: Waiting for model URDF on the ROS param server at location: //arm_controller/robot_description
Found 0 motors
Motors started.
[ INFO] [1506001038.799179925]: SquirrelHWInterface ready.
Intializing loop...
Looping...
[arm_controller/squirrel_control_node-6] process has died [pid 16855, exit code -11, cmd /home/squirrel/catkin_ws/devel/lib/squirrel_control/squirrel_hw_main __name:=squirrel_control_node __log:=/home/squirrel/.ros/log/668f2dfc-9ec6-11e7-bfb3-00190f153f00/arm_controller-squirrel_control_node-6.log].
log file: /home/squirrel/.ros/log/668f2dfc-9ec6-11e7-bfb3-00190f153f00/arm_controller-squirrel_control_node-6*.log
[arm_controller/squirrel_control_node-6] restarting process
process[arm_controller/squirrel_control_node-6]: started with pid [17168]
[ INFO] [1506001040.125660436]: Waiting for model URDF on the ROS param server at location: //arm_controller/robot_description` 
Not sure if the URDF model is the also an issue, but it seems it cannot find any motors of the arm.
- dynamixel_controllers do not start as the device /dev/ttyTilt is not available anymore.
Is pan and tilt on the raspberrypi now?

ipa-nhg, 2017-09-21:
Step by step:
- First has to be solved the problem with the network: 
  - the robotino pc has to connect automatically via Wifi. I unfortunately can't solve this remotely...
  - the raspberry has to connect to the robotino pc and using the same domain
- We have to localize where are connected all the devices and redefined all the udev rules to solve the problems 5 and 6
- The error with the bringup is because you have a mix of version, the `has_8dof` was deleted, probably your `squirrel_robotino` is new and `squirrel_common` old... depending on the setup of the non-squirrel users you are using the same repositories version or not...

Once this is solved we need a new bash configuration for all the pcs, problably also a NFS and NTP setup and configure the ssh keys to be allow to launch the nodes using machine tags. All the launch files, urdf and configuration files on robotino bringup have also to be updated...

bajo, 2017-09-21:
What we have now:
- [x] Network configuration on raspberry pi
- [] ROS network configuration

-  We connect with an ethernet cable to robotino to ifdown/ifup wlan0
- Which devices should no longer be connected to robotino?
  - [] pan
  - [] tilt
  - [] ...

- I will try to start from an up-to-date catkin_ws with `squirrel_robotino` and `squirrel_common`. I'll update the issue when I know more.

Further config (NFS, NTP, etc) has too wait until we have a replacement for the broken second pc (scrat)

ipa-nhg, 2017-09-21:
From my point of view makes more sense connect the pan-tilt and door control to the robotino or scrat pc.

bajo, 2017-09-21:
I have no idea how the pan/tilt is mechanically handled now. Do we still have the dynamixel motors? Is it something completely different? It is hard to tell from looking inside the robot.

ipa-nhg, 2017-09-21:
I took a look at the code and you should have a /dev/ttyUSB devide (serial via USB port) :
https://github.com/squirrel-project/squirrel_driver/blob/indigo_dev/squirrel_interaction/src/squirrel_interaction/board/serial_api.py#L32

bajo, 2017-09-21:
That is on the raspberry pi. So pan, tilt, head pan, door are all on the raspberry pi.
I will take out the udev rules for pan/tilt and remove the dynamixel controllers from our launch file

The arm is still on robotino, however the new control does not find any motors. Any ideas?

ipa-nhg, 2017-09-21:
yes, but should be connected to one of the other pcs , is it possible?

you should have a /dev/ttyArm, is there?

bajo, 2017-09-22:
@ipa-nhg Just checked. I have a /dev/ttyArm 
`bajo@robotino:/etc/udev/rules.d$ ls -l /dev/ttyArm /dev/ttyUSB0 
lrwxrwxrwx 1 root root         7 Sep 22 13:30 /dev/ttyArm -> ttyUSB0
crw-rw-rw- 1 root dialout 188, 0 Sep 22 13:30 /dev/ttyUSB0`

Wifi did connect after the boot again. So this seems fine now

ipa-nhg, 2017-09-22:
- do you have dialout group rights??
- Download and call the following python script https://github.com/squirrel-project/squirrel_recommender/blob/master/test/test_arm.py

bajo, 2017-09-22:
Yes, I am in the dialout group.
I cannot run the script. 
`File "test_arm.py", line 2, in <module>
    from cf.hal.dxlpro import SerialDevice
ImportError: No module named cf.hal.dxlpro
`

ipa-nhg, 2017-09-22:
did you re-install your pc?? festo usually install the cf library by default.
@lokalmatador do you know a command that can be used to find the dynamixel motors?

lokalmatador, 2017-09-22:
Can you connect to the arms using the Dynamixel Wizard? If yes, ok, if not -> hardware issue with the arm. In case of finding them, if /dev/ttyArm is there and you're in dialout, everything should theoretically be fine and you should be able to connect to the motors. If not, I have no clue for now. For the script Nadia mentioned, we never tried that one. Btw, Dynamixel wizard or dynamixel workbench (http://wiki.ros.org/dynamixel_workbench) are the only tools I know for directly connecting.

bajo, 2017-09-22:
@ipa-nhg At some point we did, however we used the images from festo to install them. Is there some location where I might find the python module?
We plan to test this on Monday with the Windows tool from dynamixel. After that we hopefully know what is going on here.

ipa-nhg, 2017-09-25:
Probably Festo installed it locally with the old arm driver... I have no idea where we can find it again. But it is better if you can use the Windows tool!

bajo, 2017-10-15:
All issues were solved during the integration week.

closed 2017-10-15 by bajo