squirrel-project/squirrel_driver #141: Error when moving motors (neck, head) short after another

bajo, 2017-10-03:
it seems to be impossible to move the motors controlled by idmind's controller board at the same time (would make sense to move neck and head at the same time) or short after another.

The resulting error is:
```[ERROR] [WallTime: 1507053938.566377] bad callback: <bound method Controller.move_neck of <__main__.Controller instance at 0x7f9203f351b8>>
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 720, in _invoke_callback
    cb(msg)
  File "/u/bajo/catkin_ws/src/squirrel_driver/squirrel_interaction/src/squirrel_interaction/board/board.py", line 77, in move_neck
    self._motor.move_to("neck", self.neck_destination)
  File "/u/bajo/catkin_ws/src/squirrel_driver/squirrel_interaction/src/squirrel_interaction/board/serial_api.py", line 58, in move_to
    response.extend(self.serial.read(5))
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)```
The first movement is executed, the second one fails without feedback for the "caller node", as it is handled via a publisher/subscriber. This is not a suitable solution.

Is it possible for idmind to implement a way to move the motors in parallel?

HPovoa, 2017-10-04:
yes, hopefully it will be possible to move the motors in parallel. I am making some changes to how i handle serial communication in our squirrel, if all goes well, I will apply these changes to the others.

bajo, 2017-10-15:
Fixed by #146 and follow up pull requests.

closed 2017-10-15 by bajo