squirrel-project/squirrel_driver #143: board.py reported positions are not reliable

bajo, 2017-10-03:
I tested the movement of the neck by sending it ~10 degree increments and printed the current and the new position before the movement is actually executed.

These are the results printed from within board.py

```
current neck position: 10
next neck position: 21
current neck position: 21
next neck position: 32
current neck position: 71
next neck position: 82
current neck position: 81
next neck position: 92
current neck position: 91
next neck position: 102
current neck position: 101
next neck position: 112
current neck position: 111
next neck position: 122
current neck position: 121
next neck position: 132
current neck position: 131
next neck position: 142
```
After the second movement the data was off by 40 degree!!!! 
After the last position I again hit the issue reported in #142 
@mzillich @ipa-nhg I guess this needs to be sorted out as soon as possible.

bajo, 2017-10-03:
Update on this issue:
Sending neck and head to 0.0 should align them. tf and rviz believe this to be true. However the current result can be seen in the attached image. 
[image]

mzillich, 2017-10-03:
so the above positions are reported wrong and also physically wrong?

bajo, 2017-10-04:
Well, kindoff both. 
It started to be wrong and when I send another 10° increase it calculated from the wrong position and moved to the newly calculated position and also reported thus position as where the neck was.
But now I send 0.0 to neck and head, the joint_states published say that they are both at 0.0 and aligned but the hardware clearly points somewhere else.

bajo, 2017-10-15:
Reported values were up to 3 degrees of from the desired goal position when we tested this last week.