squirrel-project/squirrel_driver #149: STRANDS repo no longer available

bajo, 2017-10-09:
@ipa-nhg buildbot fails with `W: Failed to fetch http://lcas.lincoln.ac.uk/repos/release/dists/trusty/main/binary-amd64/Packages  503  Service Unavailable` 

As we no longer need their repo and have our own now we should take this out of all our code.

ipa-nhg, 2017-10-09:
right, removed https://github.com/buildbot-squirrel/rosdistro-squirrel/commit/be0a34acea1cd179fb54fd3fb09e2abbe2de1dae I will restart buildbot and the builds

ipa-nhg, 2017-10-09:
we need still the server for the package `ros-indigo-classifier-srv-definitions` could I please have access to the source code to add the debian to the buildbot server??

bajo, 2017-10-10:
@ipa-nhg These are part of the v4r_ros_wrappers. The naming convention was changed for them and @jibweb is preparing a pull request to adapt our code as well.
Our repository at rwiki.acin.tuwien.ac.at/apt/v4r-release provides both v4r and v4r_ros_wrappers.