squirrel-project/squirrel_driver #154: neck_pan/tilt_link changes from before the hardware changes

bajo, 2017-10-13:
@ipa-nhg To make the viewcontroller behave correctly I had to change the direction of the calculated relative pan and tilt. I assume this is due to the orientation changes when updating the urdf file. I think the orientation of the neck_pan_link and the neck_tilt_link is rotated compared to the old links. Is this something intentionally? Might be important for others to know as well.

ipa-nhg, 2017-10-13:
I usually tried to use always the same convection, the z defines the axis of the rotation and the right hand rule the movement direction. During the test remotely I saw that the motors of the shell didn't follow the convention and  I just changed the direction of the motor https://github.com/squirrel-project/squirrel_common/commit/f1c740e0fe93def95517e7b251e73d9eb9bd4a60 , the other possibility is change the joint orientation, do you prefer this option?
Feel free to modify the URDF or let me know which changes are needed

bajo, 2017-10-15:
It was faster to change the code of the viewcontroller. Moves to the desired directions again.

closed 2017-10-15 by bajo