squirrel-project/squirrel_driver #160: Buffered commands after safety node triggered

tpatten, 2017-11-10:
If the arm stops moving because the safety node is triggered, the commands seem to buffer and are suddenly applied when the safety node is reset. Results in very quick and dangerous movements.

tpatten, 2017-11-16:
@lokalmatador @smhaller

lokalmatador, 2017-11-16:
I'll fix that tomorrow.

lokalmatador, 2017-11-17:
I added a fix that should resolve that issue. Engagement of the safety_lock_ now is check not at the beginning of write but right before sending commands to any motors - this should guarantee that any internal ros_control buffer is emptied. Also, when releasing the lock, I just set the joint_position_command_ to the current joint_position_. I can't test it before Monday, as Senka is using/calibrating/preparing the robot for pushing, maybe @tpatten can briefly have a look at it. If so, please check out my fork of squirrel_driver at https://github.com/lokalmatador/squirrel_driver/.

tpatten, 2017-11-21:
We still do not have a working arm so I haven't been able to test. As soon as we fix our problem I'll take a look.

tpatten, 2018-03-05:
I think this is fixed right? The unloading/loading solution.